neoxic/ESCape32 #1: Build system generates bad paths on Cygwin

hydra, 2023-08-08:
From the documentation it's not clear what the supported build environments are, but when I tried to use CMake on cygwin it generates paths that the windows based arm cross compiler (from GNU Arm Tools) doesn't understand.

cygwin:
```
$ cmake --version
cmake version 3.25.3
```

Snippet from a build.make file
```
/cygdrive/d/Programs/GNUArmTools/active/bin/arm-none-eabi-gcc.exe $(C_DEFINES) $(C_INCLUDES) $(C_FLAGS) -MD -MT CMakeFiles/IFLIGHT2-rev5.elf.dir/mcu/STM32G071/config.c.obj -MF CMakeFiles/IFLIGHT2-rev5.elf.dir/mcu/STM32G071/config.c.obj.d -o CMakeFiles/IFLIGHT2-rev5.elf.dir/mcu/STM32G071/config.c.obj -c /cygdrive/d/Users/Hydra/Documents/dev/projects/escape32/ESCape32/mcu/STM32G071/config.c
```

As you can see, the paths are absolute and have a `/cygdrive` prefix, which is normal for cygwin tools which understand such paths.  However the c compiler is a windows tool which usually expects paths in the form `<drive>:\...`, however the gcc c compiler on windows also accepts paths like `/<drive>/...`.

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc.exe (Arm GNU Toolchain 12.2.MPACBTI-Rel1 (Build arm-12-mpacbti.34)) 12.2.1 20230214
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

If I run the same CMake commands on msys2 (ucrt64) the path is generated as follows:

msys2:
```
$ cmake --version
cmake version 3.26.4
/d/Programs/GNUArmTools/active/bin/arm-none-eabi-gcc.exe $(C_DEFINES) $(C_INCLUDES) $(C_FLAGS) -MD -MT CMakeFiles/IFLIGHT2-rev5.elf.dir/mcu/STM32G071/config.c.obj -MF CMakeFiles/IFLIGHT2-rev5.elf.dir/mcu/STM32G071/config.c.obj.d -o CMakeFiles/IFLIGHT2-rev5.elf.dir/mcu/STM32G071/config.c.obj -c /d/Users/Hydra/Documents/dev/projects/escape32/ESCape32/mcu/STM32G071/config.c
```

Note the `/d/` prefix, which the compiler seems to understand.

I had a quick look to see if it would be possible to tell `cmake` to use relative paths, which would avoid the issue, but I was unable to find out how to do this.

To summarize:
* currently building on cygwin doesn't work.
* building on msys2 ucrt64 appears to work (but unrelated to this there are other build issues relating to libopencm3 relating to missing nvic.h files)

neoxic, 2023-08-08:
If you generate build files in Cygwin environment, you need to run the build process in the same Cygwin environment. This is how Cygwin works. You can't generate build files in Cygwin and then run `make` outside of it as this is what you're trying to do as far as I can see. Your issue is unrelated to CMake.

hydra, 2023-08-10:
> If you generate build files in Cygwin environment, you need to run the build process in the same Cygwin environment. This is how Cygwin works. You can't generate build files in Cygwin and then run [...]

I have two build environments, the cygwin one doesn't work, the msys2 ucrt64 one does.  I'm not mixing environments, the only thing common to both is the GNU Arm Tools compiler, installed in `D:\Programs\GNUArmTools\active`, which is not installed using a package manager but is installed by downloading an official .zip file from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads and extracting it, and then adding the path to it into each environment.

From my perspective, there's definately an issue with cmake in the cygwin environment generating paths that the compiler can't understand.  I'm able to build Edgetx in my cygwin environment using the same cmake and compiler, so perhaps there's some clues or settings in the edge project that can be applied to ESCape32 so that it's able to build in cygwin?

neoxic, 2023-08-12:
Sorry, I misunderstood the problem. Now I see what you mean. Yes, Cygwin is made in such a way that non-Cygwin programs, such as the ARM GNU C compiler in your case, will most likely get confused when run in the Cygwin environment. For that to work, you either need a dedicated Cygwin-aware ARM GNU Tool Chain (apparently built from source under Cygwin) or fiddle around with paths. Take a look here for some more hints:
https://cygwin.com/pipermail/cygwin/2020-April/244314.html

In a nutshell, Mingw appears to be more suited for building on Windows as you have already discovered yourself.

closed 2023-08-18 by neoxic