neoxic/ESCape32 #2: Build errors relating to libopencm3 `nvic.h` files.

hydra, 2023-08-08:
Attempting to build results in compilation errors due to missing nvic.h files.

Commands used:
```
$ cmake -B build -D LIBOPENCM3_DIR=../libopencm3 -G "Ninja" -G "Unix Makefiles"
-- The C compiler identification is GNU 12.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done (0.9s)
-- Generating done (0.6s)
-- Build files have been written to: /d/Users/Hydra/Documents/dev/projects/escape32/ESCape32/build
Hydra@DESKTOP-H1JUK44 ~/.../projects/escape32/ESCape32 (master ?3)
$ cd build
Hydra@DESKTOP-H1JUK44 ~/.../escape32/ESCape32/build (master ?3)
$ make IFLIGHT2
[ 25%] Building C object CMakeFiles/IFLIGHT2-rev5.elf.dir/mcu/STM32G071/config.c.obj
In file included from d:\users\hydra\documents\dev\projects\escape32\libopencm3\include\libopencm3\cm3\nvic.h:147,
                 from d:\users\hydra\documents\dev\projects\escape32\libopencm3\include\libopencmsis\core_cm3.h:17,
                 from D:/Users/Hydra/Documents/dev/projects/escape32/ESCape32/src/common.h:22,
                 from D:/Users/Hydra/Documents/dev/projects/escape32/ESCape32/mcu/STM32G071/config.c:20:
d:\users\hydra\documents\dev\projects\escape32\libopencm3\include\libopencm3\dispatch\nvic.h:24:17: fatal error: libopencm3/stm32/g0/nvic.h: No such file or directory
   24 | #       include <libopencm3/stm32/g0/nvic.h>
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/IFLIGHT2-rev5.elf.dir/build.make:76: CMakeFiles/IFLIGHT2-rev5.elf.dir/mcu/STM32G071/config.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:1089: CMakeFiles/IFLIGHT2-rev5.elf.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:1122: CMakeFiles/IFLIGHT2.dir/rule] Error 2
make: *** [Makefile:566: IFLIGHT2] Error 2
```

See screenshots below:

[image]

[image]

neoxic, 2023-08-08:
`nvic.h` is auto-generated in libopencm3 during the build process. Have you actually built libopencm3 exactly as described in the README?

hydra, 2023-08-08:
yeah, I missed that step.

```
$ make IFLIGHT2
[ 25%] Building C object CMakeFiles/IFLIGHT2-rev5.elf.dir/mcu/STM32G071/config.c.obj
[ 25%] Building C object CMakeFiles/IFLIGHT2-rev5.elf.dir/mcu/STM32G071/preset.c.obj
[ 25%] Building C object CMakeFiles/IFLIGHT2-rev5.elf.dir/src/cli.c.obj
[ 50%] Building C object CMakeFiles/IFLIGHT2-rev5.elf.dir/src/io.c.obj
[ 50%] Building C object CMakeFiles/IFLIGHT2-rev5.elf.dir/src/led.c.obj
[ 75%] Building C object CMakeFiles/IFLIGHT2-rev5.elf.dir/src/main.c.obj
[ 75%] Building C object CMakeFiles/IFLIGHT2-rev5.elf.dir/src/telem.c.obj
[ 75%] Building C object CMakeFiles/IFLIGHT2-rev5.elf.dir/src/util.c.obj
[100%] Linking C executable IFLIGHT2-rev5.elf
[100%] Built target IFLIGHT2-rev5.elf
[100%] Generating IFLIGHT2-rev5.bin, IFLIGHT2-rev5.hex
[100%] Built target IFLIGHT2
```

Making the build process simpler and fool-proof (i'm the fool!) is why I was trying to add libopencm3 as a dependency in the cmake and to fix the commit used for libopencm3 in my branch here:

https://github.com/hydra/ESCape32/tree/add-libopencm3-as-submodule

neoxic, 2023-08-08:
Well... Not everybody needs a local libopencm3 built automatically every time they hit `make`.😊

closed 2023-08-18 by neoxic