neoxic/ESCape32 #3: After the PWM input value over 2100, the motor stops spining

Li-Tianming, 2024-03-22:
Hardware:

   ESC: [T-MOTOR F35A 3-6S](https://store.tmotor.com/product/f35a-fpv-esc.html)
   Motor: [F20Ⅱ KV2800](https://store.tmotor.com/goods-602-F20_KV2800.html)
   Firmware: TMOTOR2-rev10.bin

Description:

  4S battery, HJ Servo Tester for PWM input. No propeller.  HJ Servo Tester‘s PWM start from 800, end to 2200. The motor start to spin at PWM 1050. Motor increase spinning speed with PWM value increase, but after PWM value over 2100, motor stop spinning. It will only start spinning after PWM input fall back to  800.
   This happens almost every time, with a very small probability, maybe one in ten, that the motor will continue to spin with PWM input over 2100.
   I have tried PWM calibration, but it seems not works. With the original BliHeli firmware,  the motor can spin with PWM value over 2100 until 2200.
   
 I am very appreciate any clues or replies.

Li-Tianming, 2024-03-22:
https://github.com/neoxic/ESCape32/blob/51b96a1642bd25c30fe483c2342c1bdc71301c61/src/io.c#L241

I think i found the reason why this happend

neoxic, 2024-03-22:
A pulse width below 800 and above 2200 is considered invalid in the firmware and will lead to disarming. Hence you need to get it back to zero throttle every time. Not sure why you need anything above 2100.

I have the same HJ servo tester, and I get the same behaviour. It's normal. Automatic servo PWM calibration works well. Make sure `throt_cal` is 1.

closed 2024-03-22 by Li-Tianming

neoxic, 2024-03-22:
Please feel free to join our Discord server:
http://support.escape32.org