neoxic/ESCape32 #6: Add full CRSF support

wimalopaan, 2024-07-23:
Over CRSF the ESC could generate a config menu,. Then it could be fully configured via the transmitter/handset via ELRS.

It would also be helpful to use CRSF then as half-duplex, so only one wire (TX on the stm32 side) would be sufficient.

With this, the ESC could be connected via the CRSF serial of the receiver, whereas the servos could be connected via pwm. 

If CRSF half-duplex is used, the full/half-duplex converter should be used on the receiver-serial.

neoxic, 2024-07-25:
I can certainly set a "feature request" tag on this, but I see several problems for now:

1) Configuration over CRSF is theoretically possible, but it will require lots of effort and digging because there isn't much documentation/examples on that. Personally, I don't have any CRSF gear either. Moreover, we don't have much flash space left for STM32F051 MCUs, so it will probably need to be made available under a build option. Overall, this increasingly looks like lots of effort vs very little demand and a narrow range of affected setups.

2) Half-duplex CRSF is not supported by receivers according to this [link](https://github.com/crsf-wg/crsf/wiki/Physical-Layer). Half-duplex would also require a request/response approach to telemetry transmission in order to avoid line collisions which doesn't seem to be part of the protocol.

neoxic, 2024-08-30:
Closing this for now...

closed 2024-08-30 by neoxic