neoxic/ESCape32 #8: build fail

opportunity13579, 2024-08-29:
I build source code but i get this message,

[  0%] Building C object CMakeFiles/AART1-rev11.elf.dir/src/main.c.obj
/home/allen/ESCape32/src/main.c:68:11: error: initializer element is not constant
 Cfg cfg = cfgdata;
           ^~~~~~~
CMakeFiles/AART1-rev11.elf.dir/build.make:145: recipe for target 'CMakeFiles/AART1-rev11.elf.dir/src/main.c.obj' failed
make[2]: *** [CMakeFiles/AART1-rev11.elf.dir/src/main.c.obj] Error 1
CMakeFiles/Makefile2:315: recipe for target 'CMakeFiles/AART1-rev11.elf.dir/all' failed
make[1]: *** [CMakeFiles/AART1-rev11.elf.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

how can i fix the problem?

thank you.

neoxic, 2024-08-29:
Make sure you have the latest ARM GNU tool chain.

opportunity13579, 2024-08-29:
is this right? (Sorry i am linux beginner)

VirtualBox:~/ESCape32$ make -v
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

opportunity13579, 2024-08-29:
i build successfully , thank you

closed 2024-08-29 by neoxic