neoxic/ESCape32 #10: I need an FW

stefanbeier, 2024-09-30:
I need a Bootloader and a FW for an AT32F421K8U7 MCU with signal line on PB4.
Were can i find it?
Now the ESC run with AM32_TEKKO_F421 1.98.

neoxic, 2024-09-30:
Signal line on PB4 is not supported on the AT32F421, sorry. You can make your ESC compatible by connecting PB4 and PA2 with a shunt.

neoxic, 2024-09-30:
https://github.com/neoxic/ESCape32/wiki

neoxic, 2024-09-30:
The shunt can be any tiny wire that connects two pins - PA2 and PB4. I would use a braid from a 16 AWG wire or something.

neoxic, 2024-09-30:
[image]

closed 2024-10-01 by neoxic

stefanbeier, 2024-10-03:
A bridge between PA2 and PB4 was unsuccessful. The signal line directly on PA2 works. Can I leave the signal line directly on PA2, or should I build a resistor into the line? I measure 120 ohms between BP4 and the signal line on the ESC connector. Is the signal from the signal line processed at another point in the ESC layout, or is it only connected to the pin on the MCU?

stefanbeier, 2024-10-03:
I installed the FW Airbot1. But I don't know if it's the right one. It makes the motor rotate and I can change the ESC settings via WiFi. Can I assume that it's the right FW? What would happen if it's not the right firmware?

neoxic, 2024-10-03:
> A bridge between PA2 and PB4 was unsuccessful. The signal line directly on PA2 works. Can I leave the signal line directly on PA2, or should I build a resistor into the line? I measure 120 ohms between BP4 and the signal line on the ESC connector.

It's advised to have a series resistor. You can see on the picture that I connected PA2 to the series resistor on the signal line (that is also connected to PB4). You can do the same to reuse the same resistor. Your shunt was unsuccessful because AIRBOT1 uses PB4 for LED control, i.e. there's a conflict. If AIRBOT1 works for, you can use NEUTRONRC2 instead (it's a generic target that doesn't use any other pins except for the mandatory ones), hence your shunt will work (there will be no conflict with PB4).

> Is the signal from the signal line processed at another point in the ESC layout, or is it only connected to the pin on the MCU?

I guess that's what you mean... AIRBOT1 uses PB4 as one of the LED control lines. Use NEUTRONRC2 instead.

neoxic, 2024-10-03:
> I installed the FW Airbot1. But I don't know if it's the right one. It makes the motor rotate and I can change the ESC settings via WiFi. Can I assume that it's the right FW? What would happen if it's not the right firmware?

If it's not the right firmware, the motor will not spin properly. But there are cases when there can be damage to the ESC if for example mosfet control logic is inverted. At present, there is only one image that can cause such damage - NEUTRONRC4. It's explicitly marked in the wiki.

stefanbeier, 2024-10-03:
Thank you very much for your help. The NEUTRONRC2 firmware also works with the bridge between PA2 and PB4. Just as you wrote. But unfortunately I don't get any voltage or current as telemetry data. Other inputs of the MCU are probably being used. The AIRBOT1 got the current value, but the voltage value was not true.

neoxic, 2024-10-03:
NEUTRONRC2 is a generic target, i.e. it doesn't have voltage/current telemetry, LED control, etc. Feel free to hop into our discord server (the link is in the wiki), post close up pictures of your ESC, and we'll figure out the rest should you need that. A new target might be needed for your ESC with a _"converted to PA"_ remark as usual.

stefanbeier, 2024-10-05:
Is the telemetry output on the Airbot1 FW also on pin PB6? It is wired like this on my ESC. But no data is received by Frsky. S.PORT is selected. But I can see telemetry data via WiFi.