neoxic/ESCape32 #15: Feature request: Propeller position

maxledfreak, 2024-11-27:
Hello,
I think it will be great to support function that in airplane mode propeller will stop turning in a devined position to avoid bricking it.

Like yge: YGE
https://www.yge.de › 2023/11PDF
YGE Propeller-Positionierung

neoxic, 2024-11-28:
There is already a motor lock (active brake) feature that can be used to prevent the propeller from turning.

maxledfreak, 2024-12-01:
@neoxic yes that ist correct. But I'm thinking that the propeller should stop in a specific position. Please have a look in the pdf file. This could be important if you are not able to use a folding prop and would like to avoid damage of your propeller during landing.

neoxic, 2024-12-01:
There is no way to stop an N-pole rotor (where N>2) in a specific position without some kind of a dedicated position sensor. In other words, a specific hardware is required for that task.

Btw, I don't see a working link to a PDF file. Please attach it to this issue.

maxledfreak, 2024-12-11:
https://www.yge.de/wp-content/uploads/2023/11/DE-Propellerpositionierung.pdf

Second try with the. Pdf file. Yes I agree with you that an additional trigger is needed. Maybe different kind of sensors could be used. But first step is to understand, if an interface is available, that could trigger the esc to stop after a small amount of time to the position where the trigger was set.

neoxic, 2024-12-12:
If by "interface" you mean a pin that for example goes high when the desired position is reached and is low otherwise, any free GPIO pin can be used for that. Implementing some simple positioning algorithm in the firmware isn't that hard if I had some testing hardware (suitable Hall sensor that can be connected to a GPIO pin at least?). If you're willing to provide that, feel free to reach out to me via Discord (the link is in the wiki).

maxledfreak, 2024-12-22:
@neoxic could you tell me your name on discord?

neoxic, 2024-12-22:
It's "escape32.org". My nickname on Discord is "senyx".

neoxic, 2025-02-22:
Ok, I'm gonna close this for now in order to prevent AI  bots from opening nonsense PRs based on feature requests.

closed 2025-02-22 by neoxic