neoxic/ESCape32 #17: Android 15

Voglberg, 2025-01-16:
Hallo, mit Android 15 bei meinem Pixel 9 kann ich zum flashen der Firmware den Button zur Auswahl der Datei nicht bedienen.

neoxic, 2025-01-16:
The same on my Mac. Close that captive portal window, open your browser and navigate to `escape32.local` or `192.168.4.1` as described here:
https://github.com/neoxic/ESCape32/wiki/WiFiLink

closed 2025-01-16 by neoxic

Voglberg, 2025-01-17:
Hallo, nicht auf dem PC sondern mit dem Handy. Mit dem Handy meine Frau ( Galaxy Note 20 mit Android 13 ) funktioniert alles. Mit meinem Pixel 9 mit Android 15 kann ich den Button Firmware nicht auswählen.

Gruß

Voglberg, 2025-01-17:
> Closed #17 <https://github.com/neoxic/ESCape32/issues/17>  as completed.
>
> 
> <https://github.com/neoxic/ESCape32/issues/17#event-15968202881> , or
> unsubscribe
> ATLTAVCNFSM6AAAAABVKRQYW2VHI2DSMVQWIX3LMV45UABCJFZXG5LFIV3GK3TUJZXXI2LGNFRWC
> 5DJN5XDWMJVHE3DQMRQGI4DQMI> .

Hallo, nicht auf dem PC, sondern mit dem Handy. Mit dem Handy meine Frau (
Galaxy Note 20 mit Android 13 ) funktioniert alles. Mit meinem Pixel 9 mit
Android 15 kann ich den Button Firmware nicht auswählen.

Gruß