neoxic/ESCape32 #21: PHOTONDRIVE1 - ext. telemetry pin

timemaster5, 2025-04-20:
Hey! We usually go with bidirectional DShot for our builds, but this time we’ve got a request to use an external telemetry pin. In our setup, the telemetry pin is hooked up to PB6 on the PHOTONDRIVE1 target. Could you take a look when you get a chance? I couldn’t find any reference to that pin in the source code, so I’m guessing it might not be set up correctly.

neoxic, 2025-04-20:
Have you actually tested telemetry on the telemetry pad before creating this issue?🙂 Because it works for me.

timemaster5, 2025-04-20:
Yes, it seems it doesn't work on Pixhawk for some reason, I don't have a Pixhawk personally, but I received this report:

neoxic, 2025-04-20:
So you tested it with Betaflight and it works, but it doesn't work with Pixhawk. It seems to me that the issue is unlikely to be on the ESC side. We've had these guys pop up in our discord first (I hope you're also on our discord), and I've already told them that ESC never sends zero or negative values (all data is unsigned for example). Since they receive telemetry, telemetry requests via DSHOT clearly work as expected. What I don't understand is why open issues without any kind of clear evidence. There can be issues in their setup, something in PIX4, or anything else really.

neoxic, 2025-04-20:
Check this out for example:
https://discord.com/channels/1022170275984457759/1362438669364891648

Looks very similar, but it's not ESCape32.

neoxic, 2025-05-05:
Any update on this? Because I can clearly see correlation between dropouts in the picture due to some external issue. It's unlikely to have telemetry values incorrect all at once with a normally running motor even in theory.

[image]

timemaster5, 2025-05-09:
Thank you, @neoxic, for all your responses. It seems the original author is no longer interested in this issue, and I agree that it is most likely a problem on "the other side."

When I initially opened this issue, I was curious whether other flight controllers might require a different approach data-wise. However, that doesn't quite make sense, as the data format is defined by the protocol in use. I was surprised by the PX4-related issue you linked, as it appears quite similar. Perhaps there's an opportunity to address the distortions electronically instead.

To make this issue more useful for others, I suggest renaming it to "Pixhawk/PX4 Telemetry Issues – Target: PHOTONDRIVE1" and closing it with the above information archived. 

That way, anyone encountering a similar problem will have a reference point for a potential fix. Once we get our hands on a Pixhawk ourselves, we can also investigate the source of the noise and explore possible hardware-level solutions.

neoxic, 2025-05-09:
Done.👍

neoxic, 2025-05-10:
Ok, this is epic! I actually stumbled across this thing from a completely different angle. There is a USB-C power delivery dead battery detection mechanism in the STM32G431 that **is enabled by default**.🤦‍♂️ Two pins (PB4 and PB6) depend on the level of two other pins (PA9 and PA10 - motor PWM pins):

[image]

How it's possible that this functionality is enabled by default is totally beyond me.

closed 2025-05-10 by neoxic

neoxic, 2025-05-10:
Please note though that the real issue may still be unrelated to the firmware. What has actually been fixed is possible interference between the telemetry pin (PB6) and one of the motor control pins (PA9) whereby incorrect signal levels could have been introduced on the telemetry line due to the aforementioned UCPD functionality.

timemaster5, 2025-05-10:
Wow! Nice catch — that could definitely be the cause. I’ll try to get someone with a Pixhawk to test it.

timemaster5, 2025-05-16:
I have a feedback that this commit https://github.com/neoxic/ESCape32/commit/803743bae858fd93a8c848caa3fa698fdd16e92c really fixed the telemetry data. 

Now they are clear, so thank you!

 It seems the temperature measurement is however lost entirely from the telemetry data stream. It is a different thing, so I will first try it myself and ask on discord if needed!

neoxic, 2025-05-16:
Temperature is part of the KISS telemetry data packet along with other data, so it can't just vanish. On the other hand, the PHOTONDRIVE1 target has external temperature sensor enabled. Hopefully, nothing has changed in the schematic, and we don't get zero temperature readings, but I can't check that since I don't have the updated board (R-150).🙂

commit 803743ba (2025-05-10): Firmware revision 14 (patch 3)

Add CRSF temperature and RPM telemetry
STM32G431 support for internal Hall XOR signal
Cosmetics

Fix #21